mccdaq/daqhats #1: Tips on installing daqhats in python venv

jacob-lapenna, 2019-02-20:
I'd like to develop inside a python 3 venv environment to prevent future broken dependencies on other libraries I am using with my RPi, however, I cannot seem to install daqhats in the environment. I successfully sudo installed to root, following instructions at https://mccdaq.github.io/daqhats/install.html, and everything works great in my home environment. With `~/daqhats/install.sh` installed as root for python 3, and `alias python=python3`, I've tried the following within the virtual environment `(venv) user:~/Projects/my_project $`:

1. `sudo ~/daqhats/install.sh`
2. `python ~/daqhats/setup.py install`
3. `python ~/daqhats/setup.py build`
4. Re-performing the instructions at https://mccdaq.github.io/daqhats/install.html within the environment.

I still keep getting the following error:

```
from daqhats import mcc118, OptionFlags, HatIDs, HatError
ImportError: cannot import name 'mcc118'
```
Deleting 'mcc118' so that 'OptionFlags' is the first encountered item in the import raises a subsequent error, and so on down the line of imports. However, if I simply have `import daqhats`, no error is thrown until I also try to import from daqhats_utils.py, at which point `ImportError: cannot import name...` is thrown based on the imports at the head of this python script.

nwright-mcc, 2019-02-21:
Try:

`pip install ~/daqhats` from within your virtual environment.

jacob-lapenna, 2019-02-21:
Yep, that worked. Thanks!

closed 2019-02-21 by jacob-lapenna