mccdaq/daqhats #2: Web server example Import Error

jacob-lapenna, 2019-02-20:
After `pip install dash dash-renderer dash-html-components dash-core-components` and running `$ ./web_server.py` or `$ python web_server.py` I get `ImportError: cannot import name Event` from `dash.dependencies`. After reading that the Event system was removed (see https://community.plot.ly/t/why-cant-i-import-the-event-dash-dependency/19288), I commented it out in the import statement and naturally got `NameError: name 'Event' is not defined` on line 389 of web_server.py with no import error. Potential guidance on a fix is detailed here: https://community.plot.ly/t/dash-events-button/19791/3?u=nedned

nwright-mcc, 2019-02-21:
Thank you for the feedback, I have forwarded this to the web server example developer so we can incorporate the fix.

nwright-mcc, 2019-02-28:
We have added instructions in the readme for the web server example on how to work around this issue.

closed 2019-02-28 by nwright-mcc