mccdaq/daqhats #3: Timeout accessing a resource

rreuscher, 2019-02-25:
Hi, i have purchased 4 MCC118 Hats and trying to read the inputs ( for the moment of 2 HATs ), using the examples in C. When I try program multi_hat_synchronous_scan. On the second hat i always get a result code -3 ( There was a timeout accessing a resource ) during the call of mcc118_a_in_scan_start.
Funny enough this does not happen all the time, but in abt 90% of all attempts.
Also the python version, aborts ( with message: IndexError: list index out of range ). When I read data from just one single HAT there are no issue ( does not matter which of the 4 installed HAT;s i choose).
What could be the issue ?

Best regards, Rene Reuscher

jeffreyg3, 2019-02-26:
Hello Rene,
We have heard from our office in China with the same problem.  Are you working with MCC China on this issue as well?

Jeff

rreuscher, 2019-02-26:
Hi Jeff, 

No I'm not working with MCC China on this. I was not aware there was also an 'MCC china'

Rene

jeffreyg3, 2019-02-26:
Hello Rene,
Good to know, we are working on it, and will get back to you soon (with a solution).  Please stand by....
Jeff

rreuscher, 2019-02-26:
Hi Jeff,

Thank you for the quick response
Rene

nwright-mcc, 2019-02-28:
This issue has been fixed in version 1.1.0.1 of the library.

closed 2019-02-28 by nwright-mcc