mccdaq/daqhats #7: libi2c-dev library conflict during installation.

pcouncil86450, 2019-05-14:
I found that if you want to install the daqhats library you have to uninstall libi2c-dev, install MCC daqhats library, then reinstall libi2c-dev. If you try to install the MCC daqhats library with libi2c-dev installed there will be errors that prevent communication with the daqhat.

My other i2c hardware works fine after this process so far. I also ran the other i2c hardware in addition to running the daqhat at the same time. 

Since others may run into this issue in the future I wanted to raise the issue so others can find the solution.

nwright-mcc, 2019-05-14:
What sort of issues do you have if you do not uninstall libi2c-dev first?

Does install.sh ask you about enabling the I2C interface?  If so, are you answering yes or no?

Thanks!

pcouncil86450, 2019-05-14:
install.sh does not ask about enabling the I2C interface since I already have enabled it prior to running install.sh for my other I2C devices.

In the log the install.sh has issues because some things have been redefined by libi2c-dev. I think this is what causes it to fail with the other steps of the installation.

Here is the log after running install.sh:

Building and installing library

make: Entering directory '/home/pi/daqhats/lib'
gcc  -I../include -I/opt/vc/include -fPIC -Wall -Wextra -g -O2 -MMD -c util.c -o build/util.o
gcc  -I../include -I/opt/vc/include -fPIC -Wall -Wextra -g -O2 -MMD -c mcc118.c -o build/mcc118.o
gcc  -I../include -I/opt/vc/include -fPIC -Wall -Wextra -g -O2 -MMD -c mcc152.c -o build/mcc152.o
gcc  -I../include -I/opt/vc/include -fPIC -Wall -Wextra -g -O2 -MMD -c mcc152_dac.c -o build/mcc152_dac.o
gcc  -I../include -I/opt/vc/include -fPIC -Wall -Wextra -g -O2 -MMD -c mcc152_dio.c -o build/mcc152_dio.o
In file included from mcc152_dio.c:12:0:
/usr/include/linux/i2c-dev.h:37:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^~~~~~~
In file included from mcc152_dio.c:11:0:
/usr/include/linux/i2c.h:68:8: note: originally defined here
 struct i2c_msg {
        ^~~~~~~
In file included from mcc152_dio.c:12:0:
/usr/include/linux/i2c-dev.h:89:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~
In file included from mcc152_dio.c:11:0:
/usr/include/linux/i2c.h:131:7: note: originally defined here
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~
makefile:30: recipe for target 'build/mcc152_dio.o' failed
make: *** [build/mcc152_dio.o] Error 1
make: Leaving directory '/home/pi/daqhats/lib'
make: Entering directory '/home/pi/daqhats/lib'
make[1]: Entering directory '/home/pi/daqhats/include'
make[1]: Leaving directory '/home/pi/daqhats/include'
install: cannot stat './build/libdaqhats.so.1.1.0.2': No such file or directory
makefile:35: recipe for target 'install' failed
make: *** [install] Error 1
make: Leaving directory '/home/pi/daqhats/lib'
make: Entering directory '/home/pi/daqhats/lib'
make: Leaving directory '/home/pi/daqhats/lib'

Building and installing tools

make: Entering directory '/home/pi/daqhats/tools'
gcc -I/usr/local/include -I../lib -g   -c -o mcc118_firmware_update.o mcc118_firmware_update.c
gcc -o mcc118_firmware_update mcc118_firmware_update.o -ldaqhats 
/usr/bin/ld: cannot find -ldaqhats
collect2: error: ld returned 1 exit status
makefile:15: recipe for target 'mcc118_firmware_update' failed
make: *** [mcc118_firmware_update] Error 1
make: Leaving directory '/home/pi/daqhats/tools'
make: Entering directory '/home/pi/daqhats/tools'
install: cannot stat 'mcc118_firmware_update': No such file or directory
makefile:25: recipe for target 'install' failed
make: *** [install] Error 1
make: Leaving directory '/home/pi/daqhats/tools'
make: Entering directory '/home/pi/daqhats/tools'
make: Leaving directory '/home/pi/daqhats/tools'

Building examples

make: Entering directory '/home/pi/daqhats/examples/c'
make -C mcc152/. all
make[1]: Entering directory '/home/pi/daqhats/examples/c/mcc152'
make -C digital_input_read_port/. all
make[2]: Entering directory '/home/pi/daqhats/examples/c/mcc152/digital_input_read_port'
gcc -c -o digital_input_read_port.o digital_input_read_port.c -Wall -I/usr/local/include -g
gcc -o digital_input_read_port digital_input_read_port.o -Wall -I/usr/local/include -g -ldaqhats
/usr/bin/ld: cannot find -ldaqhats
collect2: error: ld returned 1 exit status
makefile:14: recipe for target 'digital_input_read_port' failed
make[2]: *** [digital_input_read_port] Error 1
make[2]: Leaving directory '/home/pi/daqhats/examples/c/mcc152/digital_input_read_port'
makefile:7: recipe for target 'digital_input_read_port/.' failed
make[1]: *** [digital_input_read_port/.] Error 2
make[1]: Leaving directory '/home/pi/daqhats/examples/c/mcc152'
makefile:7: recipe for target 'mcc152/.' failed
make: *** [mcc152/.] Error 2
make: Leaving directory '/home/pi/daqhats/examples/c'

Reading DAQ HAT EEPROMs

./install.sh: line 36: daqhats_read_eeproms: command not found

Installing library for Python 3
Processing /home/pi/daqhats
Installing collected packages: daqhats
  Running setup.py install for daqhats ... done
Successfully installed daqhats-1.1.0.2

Do you want to install support for Python 2? [y/n] y
Installing library for Python 2
Processing /home/pi/daqhats
Requirement already up-to-date: enum34 in /usr/lib/python2.7/dist-packages (from daqhats==1.1.0.2)
Installing collected packages: daqhats
  Running setup.py install for daqhats ... done
Successfully installed daqhats-1.1.0.2

Install complete

nwright-mcc, 2019-05-14:
It is a conflict between the kernel headers and libi2c-dev.  I can avoid it with a change to the #includes in mcc152_dio.c, and will have the fix in the next release.

closed 2019-06-10 by nwright-mcc