mccdaq/daqhats #8: dash lib install error

florinsamoila, 2019-05-28:
"Could not find a version that satisfies the requirement install (from versions: )"
"No matching distribution found for install"

I get this when running:
pip install -Iv dash==0.30.0 dash-renderer==0.15.0 dash-html-components==0.13.2 dash-core-components==0.38.0

jperreault-mcc, 2019-05-28:
Dash version 0.43.0 removed the Event system, which is used in the web server example.  The next daqhats library release will include and update to the web server example to be compatible with the latest version of Dash.

For now, can you please try "pip install --upgrade pip" and then see if the issue still persists.

florinsamoila, 2019-05-28:
actually, I install latest dash and deleted the event import and this line #389 `[Event('timer', 'interval')]`

florinsamoila, 2019-05-28:
After that it seemed to work fine.

jperreault-mcc, 2019-05-28:
In order for the example to work properly with the latest version of Dash, you need to replace the `Event` in the `update_strip_chart_data` callback with an `Input` using `n_intervals`.  For more information, see [https://github.com/plotly/dash/issues/531](https://github.com/plotly/dash/issues/531).

florinsamoila, 2019-05-29:
Is it possible to get a continuous/real-time display of the waveform (oscilloscope like) in the html page?

jperreault-mcc, 2019-05-29:
Deleting the event import and line #389 from the example is what is preventing the display from updating. You will either need to install an older version of Dash that included the Event system and restore the lines of code you deleted or replace the Event with and Input using n_intervals.

**Option 1**:
> $ pip install --upgrade pip
> $ pip uninstall dash  dash-renderer dash-html-components dash-core-components
> $ pip install -Iv dash==0.30.0 dash-renderer==0.15.0 dash-html-components==0.13.2 dash-core-components==0.38.0

**Option 2**:
Line 26 From: `from dash.dependencies import Input, Output, State, Event`
Line 26 To: `from dash.dependencies import Input, Output, State`

Line 158 From: `interval=1000*60*60*24  # in milliseconds`
Line 158 To: ` interval=1000*60*60*24,  n_intervals=0`

Line 385 From: `[Input('status', 'children')]`
Line 385 To: `[Input('timer', 'n_intervals'), Input('status', 'children')],`

Line 391 From: `def update_strip_chart_data(acq_state, chart_data_json_str,`
Line 391 To: `def update_strip_chart_data(n_intervals, acq_state, chart_data_json_str,`

Line 389 - Delete the entire line

florinsamoila, 2019-05-30:
Hello again! I managed to get it working, thank you for your very good instructions!

Now, I have another question: which pins in the GPIO connector are not used, because I want to hookup some type of external physical control like a potentiometer.

<raspberryPi>--serial communication--<MCU>--analog input from--<pot>

florinsamoila, 2019-05-30:
A UART port would be very useful, or at least two spare GPIO pins so I can software emulate the UART.

nwright-mcc, 2019-05-30:
Refer to the electrical specification in the documentation under "Interface Specifications" for a list of the used GPIO pins for the device: [https://mccdaq.github.io/daqhats/_static/esmcc118.pdf](https://mccdaq.github.io/daqhats/_static/esmcc118.pdf).  

Please use MCC support rather than the github issues area for general questions about the device.

jeffreyg3, 2019-05-30:
From the MCC 118 data sheet (https://www.mccdaq.com/PDFs/specs/DS-MCC-118.pdf)
Raspberry Pi GPIO pins used:
GPIO 8, GPIO 9, GPIO 10, GPIO 11 (SPI interface)
ID_SD, ID_SC (ID EEPROM)
GPIO 12, GPIO 13, GPIO 26, (Board address)
Data interface type: SPI slave device, CE0 chip select
SPI mode: 1
SPI clock rate: 10 MHz, max

closed 2019-05-30 by nwright-mcc