mccdaq/daqhats #14: HatError -3 After One to Two Thousand Finite Samples

jacob-lapenna, 2019-08-20:
I want to take a one second long finite sample every few seconds for several days. The problem is that everything runs as expected for several hours until the program just quits on HatError -3. My relevant python code is as follows:
```
#start background acquisition into buffer
def acquire_data():
	channels = [0, 1, 2, 3]
	channel_mask = chan_list_to_mask(channels)
	num_channels = len(channels)
	#make sure samples and scan_rate match
	samples = 12000 # this ensures 1 Hz resolution in Fourier space
	scan_rate = 12000.0
	options = OptionFlags.DEFAULT
	hat.a_in_scan_start(channel_mask,
		samples,
		scan_rate,
		options)
	#read and store data
	read_values = hat.a_in_scan_read_numpy(samples, 1).data
	read_values = np.reshape(read_values, (-1, 4))
	hat.a_in_scan_cleanup()
        # only take channel 0, may want others in the future
	get_thd(read_values[:,0], num_channels, samples, scan_rate)
```
This function is called and looped via:
```
if __name__ == '__main__':
	acquire_data()
	while True:
		time.sleep(3)
		acquire_data()
```
The error is:
```
Traceback (most recent call last):
  File "THD_DAQ.py", line 100, in <module>
    acquire_data()
  File "THD_DAQ.py", line 88, in acquire_data
    options)
  File "/home/pi/Projects/THD/thd_env/lib/python3.7/site-packages/daqhats/mcc118.py", line 547, in a_in_scan_start
    result))
daqhats.hats.HatError: Addr 0: Incorrect response -3.
```
Tracing these errors from line 547 in mcc118.py leads to the daqhats/lib/mcc118.c. From there, my best guess is that `_spi_transfer` function is returning a value which is not `RESULT_SUCCESS` to the 'result' variable in `mcc118_a_in_scan_start` on line 1707. The `_spi_transfer` function has many possible return values other than `RESULT_SUCCESS`, is there a way I can print or see this value for debugging reasons?

nwright-mcc, 2019-08-20:
The return values are defined in daqhats.h, in the enum ResultCode.  Which version of daqhats do you have installed?  You can determine this from README.md in the daqhats directory.  If it is not 1.2.0.1 or higher, pull the latest version (`git pull` from inside your daqhats directory) and install it (`sudo ./install.sh`) then try your code again.  Let me know if you already have 1.2.0.1 and we can work on finding a solution.

jacob-lapenna, 2019-08-20:
Also, what's interesting is after the program stops on this error, trying to restart the program produces the same error. The only way to restart the program is to restart the daqhat. I actually have to shutdown the pi, unplug it so that the daqhat powers off, and plug it back in to restart the pi and the daqhat. Simply rebooting the pi still produces the same error, the daqhat must be reset in order for it to work again.

jacob-lapenna, 2019-08-20:
Yes, I purposely pulled the library this week to be sure I was up to date (version 1.2.0.1).

nwright-mcc, 2019-08-20:
Are you comfortable trying a development version of the library?  I have seen thread sync issues on another board I am working on and have not ported those changes into the 118 yet.  I can port them into a dev branch for you to try, but it would be a while before they went through our testing and were released.

jacob-lapenna, 2019-08-20:
Sure, I can install in another python venv...

nwright-mcc, 2019-08-20:
I will run some tests then let you know when you can try it.

jacob-lapenna, 2019-08-20:
Sounds good! Thanks!

nwright-mcc, 2019-08-20:
I have a test running here, but since this may take a while to reproduce you can try it as well.  There are no changes to the python code, only the shared C library changed.  You can switch to the dev branch in your existing location by running: 
`git checkout dev`  
You may need to do a `git pull` after the checkout to get the latest code.  Run `sudo ./install.sh` to build and install the changes then test your code.  You can switch back to the released code with:
`git checkout master`
Then install again.

jacob-lapenna, 2019-08-20:
Awesome, thanks. I'm running it overnight on the dev branch.

nwright-mcc, 2019-08-20:
I have added some more changes to make the variables shared between the scan thread and read functions more robust.  If you can, pull in the latest before running overnight.

jacob-lapenna, 2019-08-20:
Running on the updated dev branch now...

jacob-lapenna, 2019-08-21:
Still running...this is the longest I've seen it run without throwing the error...I'll keep it going for another 24 hours to be sure.

nwright-mcc, 2019-08-21:
Thanks, I have not seen an error either.

jacob-lapenna, 2019-08-22:
Still running. Thanks for being so fast with the code updates! I'll let you close the issue with versioning notes. Great product!

nwright-mcc, 2019-08-22:
Thanks, I will get this fix into version 1.2.0.2.  I won't close the issue until that version is released.

nwright-mcc, 2019-09-06:
This version is released now.

closed 2019-09-06 by nwright-mcc