mccdaq/daqhats #15: Data-logger write multiple channles in diffferent columns

talha74, 2019-12-12:
Hey,
I want to scan for example 3 or 4 channels with the C-Program "logger.c". But if I activate more than one channel, the values get written in one column. So there is: 
Column 1:
Channel 1 Channel 2 Channel 3
1,75754     1,4548     1,54151
1,4844       1,5895     1,54651
    ...                ...               ...

But I wan to get:
Column1     Column2
Channel 1    Channel 2
1,54161        1,78974
1,55135         1,5654
   ...                  ....

So which files do I have to modify. I found some interesting spots in "log_file.c", but ehich parts do I have to modify?

jeffreyg3, 2019-12-12:
Hello talha74,
The logger example does write one value per column when set to > 1 channel. 
What you are seeing is a country configuration issue.  In the U.S., the decimal point or dot (".") is used in floating point to denote the part of a value < 1 where in Europe et. al., the comma (",") is used.  
U.S.:
Channel 1, Channel 2, Channel 3
1.75754, 1.4548, 1,54151
Conversely, the comma is used in Excel and libra to separate one reading from another.
Europe:
1,75754 1,4548 1,54151
That is why there is a file type known as .csv (comma separated values)

You would need to adjust your system settings to comply with this, or when opening a log file remember to specify the column separator or make changes to the log_file.c (you guessed correctly).
See line 184 of that file (shown here):
sprintf(buf,"%2.6lf,", read_buf[scan_start_index+j]);

in particular, the section "%2.6lf," is the formatting of the acquired reading, 2 digits before the decimal and 6 digits after.
if you change the "," after the "lf" to a "." or colon or space it will place that character between values.
Then when you open the file, set the separator option to the same character you chose.

talha74, 2019-12-17:
Thanks. But the problem is, I cant compile the file. I keep getting the error, that the file "gtk.h" could not be found. I already installed the dependencies.

jeffreyg3, 2019-12-17:
_gtk.h is part of the GTK+ installation noted in the README.md:_

**Install required packages:**

sudo apt install libgtk-3-dev at-spi2-core autoconf libtool

**Install GTKDatabox:**
cd ~
git clone https://github.com/erikd/gtkdatabox.git
cd gtkdatabox
./autogen.sh
./configure
sudo make install
Run ldconfig after installing the dependencies.

sudo ldconfig

_Either you missed a step or something went wrong during the installation._

closed 2020-01-06 by nwright-mcc

talha74, 2020-01-27:
How can I change the comma in the values to a point. So I want to print into a column e.g. 1.4454 V instead 1,4454. 
Which line do I have to change?