mccdaq/daqhats #16: hat_error_message is missing a case for undefined

jschank, 2020-01-05:
There is an error in the source code for the lib.

hat_error_message(RESULT_UNDEFINED), will return the error message for RESULT_COMMS_FAILURE
There is a mismatch between the strings in the HAT_ERROR_MESSAGE array.

The Fix is to add another Case in hat_error_message(int result), and to increment the index in the default case.

IMO it would be better to put the undefined error message, as its own string. (and not in the array) that way, in the future, a missed index would at least return the undefined message which is slightly less wrong than what it does now. :)

nwright-mcc, 2020-01-06:
Thanks, I'll get this into the next release.

nwright-mcc, 2020-03-17:
Fix is released.

closed 2020-03-17 by nwright-mcc