mccdaq/daqhats #19: Unable to Compile/Run Data Logger Example

WorldwideBrandt, 2020-03-12:
I am unable to run this example when using the command window or Geany. When using Geany, I receive a number of errors like:

logger.c:103.5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

I am able to run the continuous example though

nwright-mcc, 2020-03-17:
Did you install all the prerequisites per the README.md in the example directory?  What version of Raspbian are you using?

jeffreyg3, 2020-03-17:
I Googled the error, the search showed:

You need to add one of the suggested options to the compile command configured in geany,

In the menu.. Build -> Set Build Commands add "-std=c99" after "gcc" in Compile and Build boxes.

WorldwideBrandt, 2020-03-22:
> Did you install all the prerequisites per the README.md in the example directory? What version of Raspbian are you using?

I did follow the readme. I am running Version 8

WorldwideBrandt, 2020-03-22:
> I Googled the error, the search showed:
> 
> You need to add one of the suggested options to the compile command configured in geany,
> 
> In the menu.. Build -> Set Build Commands add "-std=c99" after "gcc" in Compile and Build boxes.

I also found this, but it did not seem to solve the issue for me.

WorldwideBrandt, 2020-03-22:
When I attempt to run the example via the terminal I receive the error: 

logger.c:103.5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

However, when using Geany I receive the error:

usr/local/include/gtkdatabox_graph.h:37:21: fatal error: gdk/gdk.h: No such file or directory
#include <gdk/gdk.h>

nwright-mcc, 2020-03-23:
I tested building this example under Jessie (Raspbian version 8) and got the same errors.  I modified logger.c so it would build on the older version of the C compiler, so can you go to your daqhats directory and run 'git pull' to download the latest code then try building it again?  Make sure you run 'sudo apt update' before installing any packages with apt.

In geany, you would need to use the Build -> Make menu to build the example.

closed 2020-04-24 by nwright-mcc