mccdaq/daqhats #23: Create release on pypi

delijati, 2020-10-29:
It would be nice to be able to install the package directly from pypi

nwright-mcc, 2020-10-30:
daqhats is principally a C library with a thin Python wrapper.  The python portion is not usable without the library, and pypi will not build & install a C library so we do not put it in pypi.

closed 2020-10-30 by nwright-mcc

ksunden, 2022-08-01:
This is in fact false, you can create a wheel which contains the compiled code and place it on pypi in a way that is installable and usable without externally installing the library.

Many of the core (especially scientific) libraries in python are principally implemented in C (see numpy, scipy, matplotlib)

Being on pypi provides the ability to add daqhats as a dependency making it much easier to build libraries on top of it to integrate mcc hardware into things like data acquisition control layers.

jacoblapenna, 2023-03-02:
I am presently trying to develop a product that contains several daqhats. It would be amazing if one could install the whole software package from my `pyproject.toml` (and thereby install my software package and all its dependencies via a one line `pip install`). @nwright-mcc, any chance of reopening this after @ksunden's comment. I mean, contrary to why you closed this issue, it is actually possible to publish to pypi. I think you could reach a lot more users this way (if that is a thing you want to do).