mccdaq/daqhats #26: Can't select address of the stacked devices

bilal2143, 2020-12-15:
Hi, I have stacked multiple DAQ HAT on a Raspberry Pi and everything is working fine except the query of available DAQ HAT devices. I am using "select_hat_device" function to determine the address of DAQ HATs. The function works perfectly when I have a single board i.e., MCC 118 on the Pi but if I use multiple boards then I get the following error:

Addr 0: Error: No HAT devices found

The devices are stacked in the following manner: 
Address 0: MCC 118
Address 1: MCC 134
Address 2:: MCC 172

The jumpers are set and the data of the devices can be read if the addresses are passed explicitly e. g. address = 0 for MCC 118 and address = 1 for MCC 134.

Please do let me know if you need any other information.

Have a nice day

jeffreyg3, 2020-12-15:
Hello,
Please verify that you ran 'sudo daqhats_read_eeproms' after changing the stack. You can test it with the MCC DAQ HAT Manager.

If it still shows the problem, try putting a different board at address 0 and see how it works (NOTE:  run 'sudo daqhats_read_eeproms' after you change the address jumpers, etc.)

Awaiting your response.

bilal2143, 2020-12-15:
Hi, actually i am working on the device remotely and currently have no access to the hardware. However, i will ask my colleague to perform the said actions and inform you. Meanwhile, can you please guide me a bit about the time at which the measurement is taken. 
I have started the scan and also read data from the MCC device (here i mean a single borad) but apparently, the data doesn't contain the timestamp at which the measurements were taken. Can you please guide me about the possibility of getting time at which the measurements were taken? for example for MCC 118: can i somehow get the time from "a_in_scan_read" function output or i have to do it manually based on my scan rate?

jeffreyg3, 2020-12-15:
Neither the mcc 118 nor the a_in_scan_read() function have the ability to add or generate a time stamp.  The good news is  your Raspberry pi computer has a real time clock. so just take a time reading when the scan starts and apply the sampling rate interval to your readings post process.

bilal2143, 2020-12-16:
Hi Jeffrey, thanks for promt reply. I would really appreciate if you could also guide me abit about "How to distinguish values of different channels from the output of a_in_scan_read() function".

I am using continous scan and let's say i get 1000 values each time i read the buffer/ scaned values. Now i want to distinguish and dump those values into a csv file in such a way that each colum would contain the values of the respective channel. If i see the out put of  a_in_scan_read() function then i see a list of values without any header/ information that which value belongs to which channel. So how can i distinguish among the values of different channenls?

jeffreyg3, 2020-12-16:
The data is interleaved and returned in a 1D array.  Meaning, say you set up your scan for channels 0 through 3, the data will be organized like this:
Ch0, Ch1, Ch2, Ch3, Ch0, Ch1, Ch2, Ch3, Ch0,…

bilal2143, 2020-12-16:
Thanks a lot for your reply. Everything is working fine now except for the address of the stacked devices. I will get back to you as soon as my colleague is done with your suggested solution.

closed 2021-03-24 by jeffreyg3