mccdaq/daqhats #32: Dash Webserver Loading

polmdm, 2021-07-06:
I have successfully installed the mccdaq and dash. However, when launching the webserver with:

`cd ~/daqhats/examples/python/mcc128/web_server
./web_server.py
`

and surfing to: 

`http://IP_ADRESS:8080`

The webserver is stuck at "Loading...." and nothing shows up there.
[image]

The ouput of the console is:
[image]

While from terminal I got:
[image]

What I'm doing wrong?

jeffreyg3, 2021-07-06:
I am running that example as I type this response.  I can confirm it does work.  I am running the latest version of daqhats software.
To launch the app using the terminal from the web_server folder, I enter:
python3 ./web_server.py

Please try that.
Also the last line of the terminal screen shows the IP address you should used.  
[image]
[image]

polmdm, 2021-07-07:
Thank you very much for the reply, even I have all the most updated versions, or at least I think so. However, when running:
`python3 ./web_server.py`

The RPi returns:
`Traceback (most recent call last):
  File "./web_server.py", line 25, in <module>
    from dash import Dash
ModuleNotFoundError: No module named 'dash'`

jeffreyg3, 2021-07-07:
According to the traceback, you do not have dash installed.

open a terminal and enter
pip install dash

also, launching your browser and surfing to http://IP_ADRESS:8080
did not work for me, and is not in the instructions provided with the example.
you are to navigate to the http://\<host>\:8080 
where you replace \<host>\  with the IP address shown on the terminal when you launch web_server.py (for an example please see my last).

polmdm, 2021-07-20:
Hi, sorry for the delay in the answer. Actually, I have already installed Dash but still if exploiting python3 it returns the error reported above.

Yeah of course I surfed to the http://<host>:8080 printed also in the terminal but it doesn't work

jeffreyg3, 2021-07-20:
Since you are using Python 3, did you install DASH with pip3:
pip3 install dash

please try that.

polmdm, 2021-07-26:
Thank you very much, it worked!

Have a nice day

closed 2021-07-26 by polmdm