mccdaq/daqhats #33: Support MCC 172 and MyPI setup

sephiroth6, 2021-07-15:
Hello there,
I'm trying your DAQ HAT on MyPI CM3+.
After installation (sudo ./install.sh), I have 0 boards found. 

> sudo daqhats_read_eeproms
> Reading...
> Found EEPROM at address 0
> Done

> sudo daqhats_list_boards 
> 0 boards found

I also tried to update fw: 
> mcc172_firmware_update 0 ~/daqhats/tools/MCC_172.fw 
> Warning - address 0 using factory EEPROM default values
> Checking existing version...
> Device firmware version 1.01
> Do you want to continue? Press Y to continue, any other key to exit. > y
> Updating...
> 
> Finished
> Checking device...
> Warning - address 0 using factory EEPROM default values
> firmware version 1.01

Something caught up my attention: "Warning - address 0 using factory EEPROM default values"
Is it possible MyPI raspberry version have something different?
What can I do? I tried 2 boards... same result.

Best regards,
Thanks

jeffreyg3, 2021-07-15:
Let me start by stating, Measurement Computing has not tested any MCC Hats using a Raspberry Pi Compute Module, nor are there plans to.

From the github.com/mccdaq/daqhats page, Hardware Compatibility;
"The MCC DAQ HATs are compatible with all Raspberry Pi models with the **40-pin GPIO header** (not the original Pi 1 A or B with the 26-pin header.) They are generally not compatible with any other brand of Raspberry Pi HAT or add-on board that attaches to the GPIO header, or devices that use the Raspberry Pi SPI interface."
Compute modules do not include the GPIO header.

I have used Raspberry Pi(TM) models 3B, 3B+, and 4B, and have never seen that warning.
Raspberry Pi models tested and supported:  Raspberry Pi A+, B+, 2, 3 (A+, B, B+), or 4.

sephiroth6, 2021-07-16:
Thank you for your reply.
Combo EPROM + GPIO header have something different on MyPI.

BTW I got a workaround, forcing the addressing with a jumper to A0 and everything is working well.

closed 2021-07-16 by sephiroth6