mccdaq/daqhats #42: MCC152 interrupt callback registration failed due to missing permission for GPIO modification/access

MxLvtt, 2022-08-05:
When trying to setup the interrupt handling for the MCC152 DIO-board (as described in one of the examples) the library failed to register the interrupt callback method.

The biggest problem for me was that I did not get any useful error messages as to why the registration failed. Luckily though, after a bit of investigation, I found the culprit and ultimately the solution:

Problem was, that the library did not have access to modify the GPIOs in '/sys/class/gpio'. By giving necessary permission with these two commands:

```
sudo chown root:gpio /sys/class/gpio/unexport /sys/class/gpio/export
sudo chmod -R 770 /sys/class/gpio
```

the interrupt handling worked perfectly fine.

Is there an FAQ section in the documentation that I can extend? Or is there an additional step in the installation/setup process of the library that's missing in the documentation or that I simply missed, which would've prevented this from happening?

Or maybe the permission could be granted at the end of the install.sh script. That way it's guaranteed to work for those who would have had the same problem.

Here's some additional information about my setup:
Board: MCC152
Language: Python
Raspberry Pi Model: 3 (B)
Raspberry Pi OS: Raspbian (a pretty much fresh install)
Daqhats library version: 1.4.0.3

jacoblapenna, 2022-08-22:
I just encountered the same issue. @MxLvtt, thank you for documenting the workaround here.

Not sure how I feel about having the install script mess with permissions, but this should certainly be included in the documentation. Perhaps a more verbose error message could be raised from `daqhats.interrupt_callback_enable()` if permission issues are encountered. Without this open issue, the traceback I received would have likely sent my diagnostic wheels spinning for a long time.

My traceback was the following:

```
Traceback (most recent call last):
    File "/home/pi/Projects/MCC_DAQHats/MCC_152/Rotary_Encoder_WT/Rotary_Encoder_Example.py", line 64, in <module>
        encoder = Encoder()
    File "/home/pi/Projects/MCC_DAQHats/MCC_152/Rotary_Encoder_WT/Rotary_Encoder_Example.py", line 17, in __init__
        interrupt_callback_enable(self._count_rev, self.direction)
    File "/home/pi/Projects/MCC_DAQHats/MCC_152/Rotary_Encoder_WT/.env/lib/python3.10/site-packages/daqhats/hats.py", line 320, in interrupt_callback_enable
        raise Exception("Could not enable callback function.")                                                                                                                                                         Exception: Could not enable callback function.
```

nwright-mcc, 2022-08-22:
The pi user is a member of the gpio group by default, so these actions are not necessary.  When the MCC 152 library was written, you could open /sys/class/gpio/export in O_RDWR mode, but now it has to be O_WRONLY.  Version 1.4.0.5 fixes this, and the interrupt callbacks should work again.

closed 2022-08-22 by nwright-mcc