mccdaq/daqhats #43: Is the MCU programable via Atmel ICE

taherrera, 2022-08-18:
Hello,

I bought an MCC172 and it is working fine with the RPi, however I would like to program some firmware into the ATSAMG55J19 without the Rasperry Pi.

Ive been trying to flash some firmware using an ATMEL-ICE, but the device is not recognized by the debugger.

Is it possible to program the MCC172 via the pads on the back of the board? Or are the SWD pins disabled ?

nwright-mcc, 2022-08-18:
The SWD pins are not disabled and you should be able to use them.  The pinout is:
[image]

The ERASE pin is brought out to an unpopulated resistor R29.  You can erase the part to the factory state by shorting across those pads while powering on the MCC 172.

We use the ROM bootloader, so you can always program the MCC 172 back to the MCC firmware with the firmware update program included in daqhats.

taherrera, 2022-08-18:
I could not read the device due to bad contacts with the board. Thanks for the info !

closed 2022-08-18 by taherrera