mccdaq/daqhats #44: MCC 172  Can't open device, after X successful reads

delijati, 2022-09-07:
Hi,

after multiple time recording loops with [1] We run into [2] After that the device won't start anymore [3] Only multiple reboots of the system does help and also updating the firmware [4].

I assume something is not really cleaned up properly? We encountered this problem only with this board so far. 

Thanks,

[1] 
```
...
def record(self):
   self.hat.a_in_scan_start(...)
   while not self.timeout.is_timeout:
      read_result = self.hat.a_in_scan_read(...)
   self.hat.a_in_scan_stop()
   self.hat.a_in_scan_cleanup()
   ...
   return frames   
```

[2] 
```
rc[1990]: Traceback (most recent call last):
rc[1990]:   File "/rc/audio_sink/daq_hat.py", line 153, in _cleanup
rc[1990]:     self.hat.a_in_scan_stop()
rc[1990]:   File "/daqhats/mcc172.py", line 1153, in a_in_scan_stop
rc[1990]:     raise HatError(self._address, "Incorrect response.")
rc[1990]: daqhats.hats.HatError: Addr 0: Incorrect response.
```

[3]
```
Error while initializing DAQ Raspberry-Hat: Addr 0: Board not responding.

# AND if we try to list the board we get

$ pi@sounce:~ $ daqhats_list_boards
Found 1 board(s):

Address: 0
Type: MCC 172
Hardware version: 2
Name: MCC 172 IEPE HAT
Can't open device
```

[4]
```
$ pi@sounce:~ $ mcc172_firmware_update 0 MCC_172.fw 
The device at address 0 cannot be confirmed as an MCC 172.
Do you want to continue? Press Y to continue, any other key to exit. > y
Updating...
Finished
Checking device...
firmware version 1.01

$ pi@sounce:~ $ daqhats_list_boards
Found 1 board(s):

Address: 0
Type: MCC 172
Hardware version: 2
Name: MCC 172 IEPE HAT
Firmware version:   1.01

```

jeffreyg3, 2022-09-08:
Hello,
I modified the mcc 172 finite_scan.py example to loop.
[finite_scan_loop_py.txt](https://github.com/mccdaq/daqhats/files/9530010/finite_scan_loop_py.txt)
I let it iterate 5684 times, I did not see any problems at all.
After I stopped the app I relaunched it, again no problems at my 
writing this, it has iterated over 40 times now.

Please test this app.  
I am using the default parameters used in the example.  If you are using other a_in_scan_start() parameters, please test them as well, but I recommend you keep the app as close to the original as possible.  

Regards,
Jeff

delijati, 2022-10-05:
@jeffreyg3 thanks will try ... and report back ... we also experience this problem now with another daq-hat

delijati, 2022-11-21:
Hi @jeffreyg3 we "fixed" it by setting a `options = OptionFlags.DEFAULT` from previously `options = OptionFlags.CONTINUOUS` as we know how long a recording will be. I assume `OptionFlags.CONTINUOUS` overwrites the buffer and that leads to corruption of the memory or so?

jeffreyg3, 2022-11-21:
" I assume OptionFlags.CONTINUOUS overwrites the buffer and that leads to corruption of the memory or so?"
Yes, the Continuous flag does overwrite the buffer, but does not corrupt memory as long as you off load the data before it is overwritten.
I recommend you take a look at the data_logger example source.  That example runs continuously and writes the data to file in a continuous stream.