mccdaq/daqhats #48: MCC172 examples C data_logger

andyhsieh1010, 2022-11-11:
I can execute the example C "logger".

But I can't see the X-axis units.

I output "10Hz" "sine wave" through a signal generator.

Please tell me what I might be doing wrong.
Thank you
[image]

jeffreyg3, 2022-11-14:
[image]
I'm not seeing that on my Raspberry Pi 4.
My guess is you have something not installed or installed wrong for this example.
In particular that graph object, the GTKDatabox.
I recommend you uninstall your existing daqhats installation,  download and install a fresh copy of the daqhats software, review and run the instructions found in the readme.md found in the folder for that example.

andyhsieh1010, 2022-11-16:
OK!
I reinstalled all programs.
When I install GTKDatabox.
This error occurs on the terminal.
But I can't solve it.
I would like to ask what is your environment?
I am using Pi3 Model B OS: Pi OS 64bit
[image]

jeffreyg3, 2022-11-16:
I've run this on RPi 3b 32 bit, and RPi 4B 32 and 64 bit.

andyhsieh1010, 2022-11-17:
HI @jeffreyg3 
It's really a system environment problem.
I changed the board from 64bit to 32bit to solve this problem.
Thanks for your help
[image]

closed 2022-11-17 by andyhsieh1010

nwright-mcc, 2022-11-18:
This has been fixed for the 64-bit OS.