mccdaq/daqhats #49: phantom voltage spikes in buffer mcc118

nwaltz, 2022-11-21:
I'm currently trying to run a continuous scan on the mcc118, but I'm getting random voltage spikes.  I'm sampling at 1kHz and reading in a buffer of 10 samples across 4 channels.  When I use ```board.a_in_read(channel)``` to read at the same frequency, however, I no longer see these spikes, which leads me to believe it's a buffer issue.  I've attached two snippets of code; the first shows these spikes, the second does not.  The image shows the data (downsampled to 100Hz) with the spikes.  Does anybody have any suggestions for how I can address this issue? Am I doing something wrong here?

raspberry pi 4
daqhats 1.4.0.4
raspberry pi os 5.15.32

```
options = OptionFlags.CONTINUOUS
scan_rate = 1000
board = None

for entry in board_list:
  if entry.id == HatIDs.MCC_118:
      print("Board {}: MCC 118".format(entry.address))
      board = mcc118(entry.address)

board.a_in_scan_start(chan_mask, 0, scan_rate, options)
actual_scan_rate = board.a_in_scan_actual_rate(4, scan_rate)
timeout = -1

time.sleep(.3)

while (time.time() - stop_time < total_time):
      
    read_result = board.a_in_scan_read(10, timeout)
```
```
while(time.time() - stop_time < total_time):
   if (time.time() - time1000 > 0.001):
      time1000 = time.time()
      for channel in range(4):
          value[channel][i] = board.a_in_read(channel)
          i += 1
```

[image]

jeffreyg3, 2022-11-21:
Hello mwaltz,

MCC has received no other reports of this phenomenon.

I recommend you ground the inputs of all channels you are using in your apps and then run your script(s).
With the inputs grounded, do you still see the spike(s)?

Also I recommend you run the data_logger example.  Again, keep your inputs grounded. Should you need assistance building the example, please read the instructions in the readme.md.

I am not sure how to interpret your graph.  The analog inputs of the MCC 118 have a max voltage range of +/-10 VDC.  Your graph shows your spikes with an amplitude of > ~290 Volts(?) and higher.

nwaltz, 2022-11-22:
Hi jeffreyg3,

The issue was coming from a pwm signal that we were using on the raspberry pi.  After removing this signal, we have no more spikes.  If I had to guess, there were problems with the power quality on the raspberry pi. Thanks for your help.

closed 2022-11-22 by nwaltz