mccdaq/daqhats #51: Insallation issue

DrBwts, 2023-02-06:
I get as far as `sudo ./install.sh` 

which gives,

`sudo: ./install.sh: command not found`

so I used 'sudo chmod a+x insatll.sh` to make sure `sudo` was seeing the script as executable & then ran `sudo ./install.sh` 

but this now give,

`sudo: unable to execute ./install.sh: No such file or directory`

then I tried `sudo install.sh`

which gives me,

`sudo: install.sh: command not found'

any ideas?

nwright-mcc, 2023-02-06:
Make sure you are in the daqhats directory before running that command.  If you cloned daqhats into a different directory, cd into that directory then run the install command.

DrBwts, 2023-02-07:
Hi I'm running from the `daqhats` directory

nwright-mcc, 2023-02-07:
Can you please attach the output of running `ls` in your daqhats directory?

closed 2023-02-08 by nwright-mcc