mccdaq/daqhats #52: Adding channels to data logger example

DrBwts, 2023-02-08:
MCC172

I ran the `logger.c` example, it all compiled fine & I got 2 channels just fine.

I then amended the code to add 2 more channels as I have 2 hats on the Pi both with their own unique address,

`#define MAX_CHANNELS 4`

It compiles fine & the GTK GUI shows check boxes for all 4 channels but when I select either channel 3 or 4 I get the following pop up error,

`Error: An incorrect parameter was passed to the function`

It still works fine for channels 1 and 2

How do I add channels?

nwright-mcc, 2023-02-08:
The logger example is not written to work with multiple devices at one time.  You would need to write your own code (possibly using the logger as an example) that opens handles to both devices, synchronizes the acquisitions & data, and so on.

DrBwts, 2023-02-08:
Ahh yes I've just found the appropriate example for mulitple hats. Is there a away for triggering via software ie a key stricke from the keyboard?

I

nwright-mcc, 2023-02-08:
The MCC 172s need to see an edge on the trigger input when synchronizing multiple devices, but you could accomplish this by connecting an unused Pi GPIO pin to the master device trigger input with a wire then bringing that GPIO high with software.

DrBwts, 2023-02-08:
OK I'll use an Arduino for now, will 5v be OK?

nwright-mcc, 2023-02-08:
Yes, the trigger input is compatible with 3.3V or 5V logic.

DrBwts, 2023-02-08:
thanks you can close this one now

closed 2023-02-08 by nwright-mcc