mccdaq/daqhats #54: Hardware overruns?

DrBwts, 2023-04-26:
7 x MCC172 Raspberry Pi 3 B+

I'm collecting data at 25.6KHz on all 14 channels & its been running fine for days but today I cant get past 10 seconds acquistition before it throws a "Hardware overrun" error. After I've loaded all the channel data into a buffer & then send that buffer via WiFi using MQTT. So roughly a packet/second.

nwright-mcc, 2023-04-26:
The SPI bus on the Raspberry Pi is unable to keep up with continuous streaming from 7 x MCC 172s at 25.6 kHz (25.6kS/s * 2 channels * 7 devices = 358.4 kS/s aggregate).  Per the data sheet, the max aggregate sample rate is 307.2 kS/s when using multiple MCC 172s, but this can vary with the type of Raspberry Pi used and the system load.

Some options are:

- Perform finite scans to the internal memory on each MCC 172 (48 kS available on each MCC 172) then read all the data before starting another scan
- Reduce the sample rate and run continuous scans
- Reduce the channel count and run continuous scans

DrBwts, 2023-04-27:
How odd, today the setup is having no overruns at all at 25.6KHz acquisition on all channels.

closed 2023-09-29 by nwright-mcc