mccdaq/daqhats #57: I want to use the MCC118 with different sampling rates per channel.

gousekid, 2023-08-07:
Is it correct that the MCC 118 has a maximum sampling of 100kHz? 
If so, I would like to use 50khz, 10khz, 10khz and so on for the 3 channels. Is this possible?

nwright-mcc, 2023-08-07:
The max sampling rate is 100kHz, but the hardware does not support different rates for different channels.

gousekid, 2023-08-07:
Thank you for your kind answer. I have one more question. Using the python example you provided, the data I get is often duplicate (different rows, but consecutive values), isn't that an invalid data value?

nwright-mcc, 2023-08-07:
No, it is not uncommon to have consecutive samples on the same channel with the same value in a 12-bit system.  At a +/- 10V input range each bit is 4.88mV, so if the signal changes by less than that amount the ADC reading may not change.

closed 2023-09-29 by nwright-mcc