mccdaq/daqhats #58: Running MCC 128 with a BananaPi M2 Zero

bgruberg, 2023-09-15:
I am trying to run a single MCC 128 board with a BananaPi M2 Zero running  Armbian 23.05.0-trunk Bookworm. 

It seems like some aspects of the installation process are specific to Rasbian and I was wondering if you have a similar installation script for Armbian. Alternatively if you could help me understand some of the steps in the installation file so I can find the equivalent in Armbian. 

For example, the installation script uses rasp-config to activate SPI I2C communication. Instead I activated it by adding the following lines to `/boot/armbianEnv.txt` :

```
overlays=spi-spidev
param_spidev_spi_bus=0
overlays=i2c0
```

However, I am still unable to fully compile all the files during installation. Currently the compiler is stuck because it is not able to import `bcm_host.h`. This seems to be a library `libraspberrypi-dev` `raspberrypi-kernel-headers`.

Thank you very much for the help and let me know if there is any more information you might need from me. 
All the best.

nwright-mcc, 2023-09-15:
The daqhats are only supported on Raspberry Pi hardware and Raspberry Pi OS.  The library uses low-level calls directly to the Broadcom IO interface hardware (thus the reliance on bcm_host.h) that would have to be ported to another processor.

closed 2023-09-15 by nwright-mcc

bearsh, 2023-09-18:
I have this library running on a iMX8 board. So it's definitely possible... I replace the boardcom lowlevel interface with gpiod, so that it will work on almost every board.

If @nwright-mcc is interested in such a solution, I may create a pull request...

bgruberg, 2023-09-18:
@bearsh would you mind sharing your Git repository with me so I can take a look at some of the changes you made to make this work?

bearsh, 2023-09-18:
@bgruberg have a look at
- https://github.com/bearsh/daqhats/tree/libgpiod
- https://github.com/bearsh/daqhats/tree/generic

in the first branch, usage of libgpiod is possible instead of rpi specific memory mapped IOs. in the second branch (which simply continuous the first), eeprom handling can be done by a tool called eeprog. furthermore, makefiles of the examples have been converted to link against libgpiod (this part can be improved probably)

bgruberg, 2023-10-12:
@bearsh Thanks for sharing the links to your repositories. What OS were you using to run this? I am trying to get this to run on Armbian.

bearsh, 2023-10-12:
I had it running on RaspiOS and Debian (in a docker container running on a
Yocto based OS). It should also run on Armbian...

I can also share my dockerfile if you are interested (once I have access to
my computer).

bgruberg, 2023-10-18:
@bearsh During the installation I get the following error: 

gpio.c:18:10: fatal error: bcm_host.h: No such file or directory
   18 | #include "bcm_host.h"
   
Which seems to be a file associated with librasperberrypi-dev and raspberry-kernel-headers which seem to be Rasbian specific files. How did you get around this?

bearsh, 2023-10-19:
@bgruberg, you need to enable libgpiod and you alos need to define the pins. something like the following should work but you need to adjust the IO definitions. the first number is the gpiochip, the second the pin on that chip.

```shell
make GPIOD=1 EXTRA_CFLAGS="-DGENERIC_HAT -DADDR0_GPIO=0,19 -DADDR1_GPIO=0,16 -DADDR2_GPIO=0,22 -DRESET_GPIO=0,13 -DIRQ_GPIO=4,1 -DIRQ_IOEXP_GPIO=4,2" -C lib all
```