mccdaq/daqhats #62: Bug with Continuous scanning with external clock

shaunmccance, 2023-11-13:
There appears to be a small bug when using hats in a synchronous setup with CLK lines connected, PY libraries

Whilst a master can be set to continuous scanning (in which the buffer size is irrelevant - assuming data is read quickly enough) this does not carry over to adjacent devices set to EXTCLOCK.  

EXTCLOCK devices seem to be limited to a finite scan set to the buffer size limit regardless of data in the buffer, and will halt at this limit regardless.  This behavior is present in the example file, changing the Options to continuous rather than trigger, the scan is halted at 10000 samples due to the state of the EXTCLOCK device

Unable to figure out a neat workaround for this other than a continuous check and restart.

nwright-mcc, 2023-11-17:
Which daqhat device(s) (MCC 118, 128, 172) are you using?

shaunmccance, 2023-11-28:
> Which daqhat device(s) (MCC 118, 128, 172) are you using?

Tis was for the MCC 128 ADC

nwright-mcc, 2023-11-28:
This appears to work as expected.  Make sure to modify the _options_ list in the multiple_hats_synchronous_scan.py example as follows to enable continuous mode on both devices:
```python
   # Define the options for each HAT device
    options = [
        OptionFlags.EXTTRIGGER | OptionFlags.CONTINUOUS,  # options for device 0
        OptionFlags.EXTCLOCK   | OptionFlags.CONTINUOUS   # options for device 1
    ]
```

shaunmccance, 2023-12-03:
Thanks for the clarification. 
Can I suggest that it may be worth adding the use of pipe union notation for combining flags to the documentation or example code if it is not there. I was missing this in the setup as having tried supplying iterables this raised errors.

nwright-mcc, 2023-12-04:
It is documented in the method parameters, but I will try to add an example to the usage.
[image]

closed 2023-12-04 by nwright-mcc