mccdaq/daqhats #64: Problem installing from ./setup.sh

daleamon, 2023-11-28:
```
gcc -o continuous_scan continuous_scan.o -Wall -I/usr/local/include -g -ldaqhats
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `exp'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `log'

```
This is on Bookworm on an RP4b that is only a few months old and is presenting itself as an arm64. The daqhats is from a freshly downloaded tar.gz, just hours ago.

nwright-mcc, 2023-11-28:
I am not able to reproduce this on a 4B with a fresh install of Bookworm 64-bit.  Can you give me more info on the command you used to download the tgz?

daleamon, 2023-11-28:
I used rpi-imager to create the SDCard.

```
```
$ cat os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
The hardware is:

$ cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03115
Serial		: 100000006174d71c
Model		: Raspberry Pi 4 Model B Rev 1.5

```
I installed this download from github.
-rw-rw-r--   1 amon amon  16559166 Nov 27 16:51  daqhats-1.4.0.8.tar.gz

I built it with your ./install.sh.

JFYI: I've been using the MCC118's as a product solution almost since they were first released and have purchased dozens of them over the years. That number might go way up next year...

Also, the first thing I do with a new MCC libdaqhats release is to debianize it. I have not done so with this release for obvious reasons, but I would be happy to share that packaging with you.

nwright-mcc, 2023-11-28:
Do you know which Raspberry Pi OS image you chose in imager (full 64-bit, full 32-bit, lite, etc?)

daleamon, 2023-11-28:
```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

```

daleamon, 2023-11-28:
I would add that since the CPU type is  showing up as arm64, it must be the 64 bit image.

daleamon, 2023-11-28:
That makes me think... they have done something with the device memory space and it is different between 32 and 64 bit images. That's why I upgraded from the older 2018 epoch library. I was hoping it would have the fixes for this addressing issue.

daleamon, 2023-11-28:
I'll bet the fix here will be relevant to the RP5. I'm intending to switch all our stuff over to it in Q1. We've got a unit for the lab on order for testing.

nwright-mcc, 2023-11-28:
Here are the commands I use and the results I get from a fresh install of Raspberry Pi OS 64-bit using rpi-imager 1.8.1 on a Pi 4B.  I also tried it with 64-bit lite with no errors.  

I did not run any commands prior to this after the first boot of the image.  The error messages you got imply that it is not linking with the math and bcm-host library (or they are not present.)  Do you seen any warnings earlier in the build when it is building the library?

```sh
pi@dev-pi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
pi@dev-pi:~ $ wget https://github.com/mccdaq/daqhats/archive/refs/tags/v1.4.0.8.tar.gz
--2023-11-28 15:22:12--  https://github.com/mccdaq/daqhats/archive/refs/tags/v1.4.0.8.tar.gz
Resolving github.com (github.com)... 140.82.112.3
Connecting to github.com (github.com)|140.82.112.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/mccdaq/daqhats/tar.gz/refs/tags/v1.4.0.8 [following]
--2023-11-28 15:22:12--  https://codeload.github.com/mccdaq/daqhats/tar.gz/refs/tags/v1.4.0.8
Resolving codeload.github.com (codeload.github.com)... 140.82.114.10
Connecting to codeload.github.com (codeload.github.com)|140.82.114.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.4.0.8.tar.gz’

v1.4.0.8.tar.gz              [      <=>                          ]  15.79M  15.0MB/s    in 1.1s

2023-11-28 15:22:13 (15.0 MB/s) - ‘v1.4.0.8.tar.gz’ saved [16559166]

pi@dev-pi:~ $ tar zxvf v1.4.0.8.tar.gz
daqhats-1.4.0.8/
daqhats-1.4.0.8/.gitattributes
daqhats-1.4.0.8/.gitignore
daqhats-1.4.0.8/LICENSE
daqhats-1.4.0.8/README.md
daqhats-1.4.0.8/daqhats/
daqhats-1.4.0.8/daqhats/.gitignore
daqhats-1.4.0.8/daqhats/__init__.py
daqhats-1.4.0.8/daqhats/hats.py
daqhats-1.4.0.8/daqhats/mcc118.py
daqhats-1.4.0.8/daqhats/mcc128.py
daqhats-1.4.0.8/daqhats/mcc134.py
daqhats-1.4.0.8/daqhats/mcc152.py
daqhats-1.4.0.8/daqhats/mcc172.py
daqhats-1.4.0.8/daqhats/pylintrc
daqhats-1.4.0.8/docs/
daqhats-1.4.0.8/docs/.gitignore
daqhats-1.4.0.8/docs/.nojekyll
daqhats-1.4.0.8/docs/_images/
daqhats-1.4.0.8/docs/_images/a0.png
daqhats-1.4.0.8/docs/_images/a1.png
daqhats-1.4.0.8/docs/_images/a2.png
daqhats-1.4.0.8/docs/_images/a3.png
daqhats-1.4.0.8/docs/_images/a4.png
daqhats-1.4.0.8/docs/_images/a5.png
daqhats-1.4.0.8/docs/_images/a6.png
daqhats-1.4.0.8/docs/_images/a7.png
daqhats-1.4.0.8/docs/_images/connector-web.jpg
daqhats-1.4.0.8/docs/_images/mcc118-clock-web.png
daqhats-1.4.0.8/docs/_images/mcc118-diag-web.jpg
daqhats-1.4.0.8/docs/_images/mcc118-web.jpg
daqhats-1.4.0.8/docs/_images/mcc118oem-web.jpg
daqhats-1.4.0.8/docs/_images/mcc128-clock.png
daqhats-1.4.0.8/docs/_images/mcc128-diag.png
daqhats-1.4.0.8/docs/_images/mcc128_diff.jpg
daqhats-1.4.0.8/docs/_images/mcc128_se.jpg
daqhats-1.4.0.8/docs/_images/mcc128oem-web.jpg
daqhats-1.4.0.8/docs/_images/mcc134-diag-web.jpg
daqhats-1.4.0.8/docs/_images/mcc134-web.jpg
daqhats-1.4.0.8/docs/_images/mcc152-diag-web.jpg
daqhats-1.4.0.8/docs/_images/mcc152-resistor-example-web.png
daqhats-1.4.0.8/docs/_images/mcc152-web.jpg
daqhats-1.4.0.8/docs/_images/mcc172-diag-web.jpg
daqhats-1.4.0.8/docs/_images/mcc172-web.jpg
[... 669 lines cut ...]

Reading DAQ HAT EEPROMs

Reading...
Found EEPROM at address 0
Found EEPROM at address 1
Done

Installing library for Python 3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/daqhats-1.4.0.8
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: daqhats
  Building wheel for daqhats (setup.py) ... done
  Created wheel for daqhats: filename=daqhats-1.4.0.8-py3-none-any.whl size=42289 sha256=b3db8c1e17235f976459514d6236f56c01cfde1e9409ae8a97d2bb051fb62b35
  Stored in directory: /root/.cache/pip/wheels/f5/d5/35/32dc86a94203645aeb64643df51bd9f5af3d4b388d68968b4b
Successfully built daqhats
Installing collected packages: daqhats
Successfully installed daqhats-1.4.0.8
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

Do you want to install support for Python 2? [y/n] n

Install complete
```

daleamon, 2023-11-28:
Thanks. Will get back to you in an hour. I've got to run our engineering telecon in 15 minutes. :-)

daleamon, 2023-11-28:
Just got handed the RP5 unit! I'll start testing it after we get this sorted. So, I'm going to do the ./install.sh and give you the output.

nwright-mcc, 2023-11-28:
The library does not currently work with the Pi 5 (see the other issues) due to how they changed the GPIO to use a new chip.  I hope to have that resolved before too long, but there is not much example code for the RP1 chip yet so it is trial and error.

daleamon, 2023-11-28:
Fortuneatly the RP5 is not immediately critical, unlike the current problem! It does appear that some things are not installed and I do not know why that would be different between your test and mine. I certainly did not delete any packages intentionally. I did run dselect to update after booting the new SDCard, and I installed developer tools and gnustep packages. None of those would be relevant to gpio.

nwright-mcc, 2023-11-28:
I thought it may be due to how you are installing from ~/Work/libdaqhats2, but I tried that here and it still works correctly.  Your issue is not related to gpio, that is just for the Pi 5.  

libdaqhats.so gets installed to /usr/local/lib, but the commands that fail seem to be linking to /usr/lib.  The library path is set by the system rather than the daqhats makefiles.  Can you check your environment variable LD_LIBRARY_PATH and the file(s) that get loaded by /etc/ld.so.conf to see if /usr/local/lib is being excluded?  For reference, the Bookworm 64-bit /etc/ld.so.conf includes /etc/ld.so.conf.d/*.conf.  In that directory, libc.conf has the contents:
```sh
# libc default configuration
/usr/local/lib
```
Maybe something modified the linker config in your system.

daleamon, 2023-11-28:
$ cat  /etc/ld.so.conf
include /etc/ld.so.conf.d/*.conf

$ ls /etc/ld.so.conf.d/
aarch64-linux-gnu.conf  fakeroot-aarch64-linux-gnu.conf  libc.conf

$ cat  /etc/ld.so.conf.d/libc.conf 
# libc default configuration
/usr/local/lib

as user shipslog:
$ printenv | grep PATH
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

$ sudo printenv | grep PATH
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

This looks like a rather thin set of  PATH definitions.

daleamon, 2023-11-28:
```
$ sudo ldconfig -v
ldconfig: Can't stat /usr/local/lib/aarch64-linux-gnu: No such file or directory
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
(from /etc/ld.so.conf.d/aarch64-linux-gnu.conf:4 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/lib/aarch64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/usr/lib' given more than once
(from <builtin>:0 and <builtin>:0)
/lib/aarch64-linux-gnu: (from /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
	libbcm_host.so.0 -> libbcm_host.so.0
	libnssckbi.so -> libnssckbi.so
	libsemanage.so.2 -> libsemanage.so.2
	libargon2.so.1 -> libargon2.so.1
	libcrypto.so.1.1 -> libcrypto.so.1.1
	libdebug_sym.so.0 -> libdebug_sym.so.0
	libavformat.so.59 -> libavformat.so.59.27.100
	libIlmThread-3_1.so.30 -> libIlmThread-3_1.so.30.5.1
	libopcodes-2.40-system.so -> libopcodes-2.40-system.so
	libdebconfclient.so.0 -> libdebconfclient.so.0.0.0
	libcupsimage.so.2 -> libcupsimage.so.2
	libssl.so.1.1 -> libssl.so.1.1
	libmariadb.so.3 -> libmariadb.so.3
	libply-splash-graphics.so.5 -> libply-splash-graphics.so.5.0.0
	libx264.so.164 -> libx264.so.164
	libliveMedia.so.77 -> libliveMedia.so.77.0.3
	libgtk-3.so.0 -> libgtk-3.so.0.2406.32
	libform.so.6 -> libform.so.6.4
	libjansson.so.4 -> libjansson.so.4.14.0
	libgdbm_compat.so.4 -> libgdbm_compat.so.4.0.0
	libwoff2dec.so.1.0.2 -> libwoff2dec.so.1.0.2
	libyuv.so.0 -> libyuv.so.0.0.1857
	libabsl_debugging_internal.so.20220623 -> libabsl_debugging_internal.so.20220623.0.0
	libgstphotography-1.0.so.0 -> libgstphotography-1.0.so.0.2200.0
	libabsl_random_internal_randen_hwaes.so.20220623 -> libabsl_random_internal_randen_hwaes.so.20220623.0.0
	librtmp.so.1 -> librtmp.so.1
	libjacknet.so.0 -> libjacknet.so.0.1.0
	liblzo2.so.2 -> liblzo2.so.2.0.0
	liblua5.3.so.0 -> liblua5.3.so.0.0.0
	libbabeltrace-ctf-metadata.so.1 -> libbabeltrace-ctf-metadata.so.1.0.0
	libibverbs.so.1 -> libibverbs.so.1.14.44.0
	libpulse-simple.so.0 -> libpulse-simple.so.0.1.1
	libjack.so.0 -> libjack.so.0.1.0
	libgcr-ui-3.so.1 -> libgcr-ui-3.so.1.0.0
	libpopt.so.0 -> libpopt.so.0.0.2
	libplds4.so -> libplds4.so
ldconfig: /lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 is the dynamic linker, ignoring

	ld-linux-aarch64.so.1 -> ld-linux-aarch64.so.1
	libubsan.so.1 -> libubsan.so.1.0.0
	libexpat.so.1 -> libexpat.so.1.8.10
	libsmime3.so -> libsmime3.so
	libnorm.so.1 -> libnorm.so.1
	libjbig.so.0 -> libjbig.so.0
	libsmbconf.so.0 -> libsmbconf.so.0.0.1
	libm.so.6 -> libm.so.6
	libarchive.so.13 -> libarchive.so.13.6.2
	libmnl.so.0 -> libmnl.so.0.2.0
	libnssdbm3.so -> libnssdbm3.so
	libhunspell-1.7.so.0 -> libhunspell-1.7.so.0.0.1
	libossp-uuid++.so.16 -> libossp-uuid++.so.16.0.22
	libLerc.so.4 -> libLerc.so.4
	libmlx5.so.1 -> libmlx5.so.1.24.44.0
	libgstbasecamerabinsrc-1.0.so.0 -> libgstbasecamerabinsrc-1.0.so.0.2200.0
	libgcc_s.so.1 -> libgcc_s.so.1
	libwoff2common.so.1.0.2 -> libwoff2common.so.1.0.2
	libffi.so.8 -> libffi.so.8.1.2
	libgstaudio-1.0.so.0 -> libgstaudio-1.0.so.0.2200.0
[... 946 lines cut ...]
/usr/local/lib: (from /etc/ld.so.conf.d/libc.conf:2)
	libdaqhats.so.1 -> libdaqhats.so.1.4.0.8
/lib: (from <builtin>:0)
	libshipscfg.so.0 -> libshipscfg.so.0.0
	libshipstypes.so.0 -> libshipstypes.so.0.0
	libBLT.2.5.so.8.6 -> libBLT.2.5.so.8.6
	libBLTlite.2.5.so.8.6 -> libBLTlite.2.5.so.8.6
	libresid-builder.so.0 -> libresid-builder.so.0.0.1
	libpigpiod_if2.so.1 -> libpigpiod_if2.so.1
	libshipstree.so.0 -> libshipstree.so.0.0
	libhardsid-builder.so.0 -> libhardsid-builder.so.0.0.1
	libshipslog.so.0 -> libshipslog.so.0.0
	libshipsprocess.so.0 -> libshipsprocess.so.0.0
	libRTIMULib.so.7 -> libRTIMULib.so.7.2.1
	libdmautils.so.0 -> libdmautils.so.0.0
	libpigpio.so.1 -> libpigpio.so.1
	libdaqhats.so.1 -> libdaqhats.so.1.3.0.6
	libsidplay2.so.1 -> libsidplay2.so.1.0.1
	libgnustep-base.so.1.28 -> libgnustep-base.so.1.28.0
	libshipscomms.so.0 -> libshipscomms.so.0.0
	libshipsensors.so.0 -> libshipsensors.so.0.0
	libobjc-qa.so.0 -> libobjc-qa.so.0.0
	libpigpiod_if.so.1 -> libpigpiod_if.so.1
 
```

daleamon, 2023-11-28:
Note that a bunch of our own product libraries are also there, all the ones prefixed as 'libships'

nwright-mcc, 2023-11-28:
Everything looks valid.  Do you still get the error if you run make in the daqhats/tools directory (in case ldconfig updated the cache)?

daleamon, 2023-11-28:
FYI:

nwright-mcc, 2023-11-28:
One test to run would be to uninstall daqhats then modify this line in daqhats/lib/makefile:
```
INSTALL_DIR = /usr/local/lib
```
changing it to /usr/lib or /lib.  Then run install again and see if the builds succeed with the library in a different location.

daleamon, 2023-11-28:
```

~/Work/libdaqhats2/tools $ make
gcc -c -o mcc118_firmware_update.o mcc118_firmware_update.c -I../include -I../lib -g
gcc -o mcc118_firmware_update mcc118_firmware_update.o -ldaqhats
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `exp'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../../lib/libdaqhats.so: undefined reference to `log'
collect2: error: ld returned 1 exit status
make: *** [makefile:23: mcc118_firmware_update] Error 1

```

daleamon, 2023-11-28:
I'm doing that and also the initial cut at a debian package I was working on before trying the raw ./install.sh (I was getting the same error there, which is why I went back to 'bare metal'.

daleamon, 2023-11-28:
Interesting. Your script runs most of the way now:

```
Successfully built daqhats
Installing collected packages: daqhats
Successfully installed daqhats-1.4.0.8
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

Do you want to install support for Python 2? [y/n] y
Installing library for Python 2
./install.sh: line 71: pip2: command not found

Install complete

```

daleamon, 2023-11-28:
I find this curious, but that gives me a base to go back to it... when I do my debian packages, they install in the standard directories rather than */local/ dirs.

daleamon, 2023-11-28:
So it looks like something between ./install.sh and Debian utils causes confusion.

daleamon, 2023-11-28:
That gives me a good starting point to track it down. I'm sure you'll want to know what I find and I'd be happy to share it.

nwright-mcc, 2023-11-28:
For that last error, Python 2 has been deprecated and the installer seems to not be available any longer.  I will be removing the Python build as part of the install in a future version anyway (it will be installed with pip, or you can manually install with setup.py.)  

Yes, let me know if there is anything I should change in the library to make it more flexible.

nwright-mcc, 2023-11-29:
Certainly, I’m happy to help.

daleamon, 2023-11-30:
I'm well along in debianizing the library. I notice that very few of the examples are built and some seem to have gone away entirely in the mcc172:

daleamon, 2023-11-30:
Incidentally, kudos on cleaning up the directory structure. It was a mess and a headache to deal with before.

daleamon, 2023-11-30:
Is /usr/local/bin/daqhats_check_152 still hardwired into the utilities? For a native debian package I'm putting it into /usr/bin and hardlnking the name to that location. If this has been cleaned up, I can drop the hardlink.

nwright-mcc, 2023-11-30:
It is still used by `daqhats_read_eeproms`, but that may change in the future.

daleamon, 2023-11-30:
We can make this a really clean drop in addition if we can figure out a way to accommodate a slight change in your lib/makefile:

# DMA20231129 changed ldconfig to point at build directory so we do not need
# to be root.
# Removed '@cd ../include; make install; cd ../lib' line because I want these
# files to appear only in the -dev package.
#
install:
        # @cd ../include; make install; cd ../lib
        @cd ../lib
        @install -d $(INSTALL_DIR)
        @install $(BUILD_DIR)/$(TARGET_LIB) $(INSTALL_DIR)
        @ldconfig -n $(DESTDIR)/$(INSTALL_DIR)
        @ln -frs $(INSTALL_DIR)/$(TARGET_LIB) $(INSTALL_DIR)/lib$(NAME).so

# DMA20231129 changed ldconfig to point at build directory so we do not need
# to be root.
uninstall:
        @rm -rf $(INSTALL_DIR)/lib$(NAME).so*
        @ldconfig  -n $(DESTDIR)/$(INSTALL_DIR)
        @cd ../include; make uninstall; cd ../lib
        @rm -rf build

nwright-mcc, 2023-12-01:
We only distribute the source so I have to keep the include install in my makefile.

nwright-mcc, 2023-12-05:
If you think they would be useful to someone fork the repo and add them (plus documentation like a readme.md) to a utils directory or similar then do a pull request and I will review it.

I will close this issue now.  Reopen it if needed.

closed 2023-12-05 by nwright-mcc

daleamon, 2024-01-31:
What is the status of the port to the RP5? I've got one on our lab bench that I'm dying to test. It could solve a large number of problems for us due to it being much faster.

Closed #64<https://github.com/mccdaq/daqhats/issues/64> as completed.

nwright-mcc, 2024-01-31:
Feel free to try the pi5 branch - everything should work, we just have not finished regression testing with older hardware and OS versions.  Please let me know if you find any issues.

daleamon, 2024-02-01:
Thanks. That's on my todo list now