mccdaq/daqhats #66: I2C 1 Not Detected

jacoblapenna, 2024-02-29:
I have an MCC 152 connected to my CM4 on the stock CM4 I/O board. Daqhats version 1.4.0.8 on RPi OS bookworm (lite version). I can run `daqhats_list_boards` and I get:

```
Found 1 board(s):
Address: 0
Type: MCC 152
Hardware version: 2
Name: MCC 152 Analog Output / Digital I/O HAT
```

Similarly, `sudo daqhats_read_eeproms` returns:

```
Reading...
Found EEPROM at address 0
Done 
```

However, upon attempting to create an `mcc152()` instance in Python, I get `daqhats.hats.HatError: Addr 0: Board not responding.`. This led me to run `i2cdetect -y 1`, which returns:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- 
```

This command returns very slowly as if a timeout is being reached for every address scanned. With multiple hats, nothing is detected. The scope trace of GPIO 2 and GPIO 3 show constant low as if this hat is holding them to ground.

When I replace the MCC152 with another, everything works. Are there test points I can look at to diagnose why this particular board is compromising the whole I2C bus? Perhaps you've seen this behavior before? Is this MCC 152 bricked? Is there another forum I should take hardware issues to?

Let me know what other information you might need from me.

nwright-mcc, 2024-03-01:
The best place for a question like this is the [Digilent MCC Forum](https://forum.digilent.com/forum/39-measurement-computing-mcc/).  

Ensure that the DIO voltage jumper W3 is installed.  

GPIO2 / 3 connect to the I/O expander used on the MCC 152 and should only be constantly low if the expander has been damaged.

jacoblapenna, 2024-03-02:
I've put a question in on the linked forum. I may of had wiring or trigger issues on the scope. It appears like normal I2C traffic on GPIO 2/3 pins though still not detecting the HAT. Also, after probing the HAT, it does not look like GPIO 2/3 go straight to the DIO expander but instead pass through an unknown 8-pin IC and a resistor before reaching the expander. If so, this could also be a point of failure.

Anyway, you can close this issue as its now posted on the Digilent forum [here](https://forum.digilent.com/topic/28404-mcc152-i2c-not-responding/#comment-86012).

nwright-mcc, 2024-03-04:
Yes, I2C goes through a logic level converter to bring it to the correct voltage for the IO expander (which depends on the position of W3) so that is also a possible point of failure.  I would try disconnecting anything on the digital I/O pins then powering up and testing the 152 with W3 in each position.  If it does not work (and other 152s set to address 0 do work) then it is most likely hardware damage.

closed 2024-03-04 by nwright-mcc