mccdaq/daqhats #69: No module named 'daqhats_utils'

PhantomPickle, 2024-06-13:
Apologies if I've missed something. I went through the instructions for installing the daqhats library and am now trying to run one of the example scripts for the MCC 172 but am getting an error when trying to import 'daqhats_utils'. I installed the daqhats module via pip as was stated in the instructions but I didn't see anything about daqhats_utils. Where can I find this module?

nwright-mcc, 2024-06-14:
daqhat_utils is a helper file that is included with the examples and is not part of the library.  When installing in a venv, you can run the examples in this manner:

```
cd ~/daqhats/examples/python/mcc128
<path_to_venv>/bin/python finite_scan.py
```

PhantomPickle, 2024-06-14:
Ahh okay, thank you! That worked.

closed 2024-06-14 by PhantomPickle